daid/LegacyCura #1353: Expert window is not visible

geeksam, 2015-10-29:
Running 15.04.2 on Yosemite/OSX 10.10.5, I can press Cmd+E all I want, but the Expert window will not appear.

geeksam, 2015-10-29:
Correction:  after I reported the issue, I got about eight Expert windows.  They appeared behind the main Cura app window and I noticed them after I moved the main window around.  After I dismissed them, I pressed Cmd+E, and the Expert window did not reappear.

Actually, I wonder whether this might be related to another issue I'm having, which is that the Print window takes a long time (like, 20 seconds or more) to appear.

daid, 2015-10-29:
OSX with Cura is always having some odd dialog issues. It's related to the wxWidgets toolkit that is being used for Cura. We're working on a full replacement of the UI made with QT, to solve these issues once and for all.

Betas of this new version are available at:
http://software.ultimaker.com/Cura_closed_beta/15.10/

geeksam, 2015-10-30:
Thanks. I'll give that a shot soon.

geeksam, 2015-11-04:
Finally got around to trying the beta.  The "Add Printer" dialog doesn't include the Printrbot, so...
¯\_(ツ)_/¯

nallath, 2015-11-05:
The people from printrbot haven't contributed profiles yet. So if they don't make it, someone from the community could / should do it.