daid/LegacyCura #1354: Can't change support settings on OS X

geeksam, 2015-10-29:
Clicking "Support Type" and "Platform Adhesion Type" sometimes changes the highlighting of the down arrow to the right of the control, but the select options do not appear.  Tabbing between controls, it takes three tabs to get from "Printing temperature" to "Diameter", but pressing Space has no more effect than clicking on the controls.

geeksam, 2015-10-29:
Update: while the text box did show "None" and the layer view didn't show any support grids on lower layers, the part I was printing did come out with them.

daid, 2015-10-29:
Just as the dialog issue, this is related to OSX and wxWidgets. Usually the dropdowns start to work when you change the size of the sidebar (you can drag it bigger and smaller)

geeksam, 2015-10-30:
Thanks for the tip.  Started behaving after I killed & restarted Cura, but I'll try that next time.

rgillan, 2016-08-31:
When is this going to be fixed?  It sometimes takes 7 or 8 reloads and dragging of the window resize bar to have any of the pull down menus active.  I'm using OSX 10.10.5.  It should be noted that the mouseover tips do not come up when it's broken like this.  I'm also using Cura 15.04.6 for another printer we have and it has no issues on OSX whatsoever (so likely something with your customisation/rebranding of Cura).  Almost makes the printer unusable.

daid, 2016-08-31:
> (so likely something with your customisation/rebranding of Cura)
Excuse me? I hope you know that I made 90% of the Cura 15.04 code. So most likely the other rebranding fixed something without contributing it back.

closed 2016-08-31 by daid

rgillan, 2016-08-31:
Sorry, didn't mean to offend. I actually thought this was a CreatBot specific queue (my very bad).  It's their changes that are broken, not Cura (which I love and use for another printer without issue).

Apologies
Rob

daid, 2016-08-31:
>I actually thought this was a CreatBot specific queue

It's not. It's not the first time a user of a re-branded Cura comes here to complain about things we didn't cause. So sorry that I'm so easily offended.

rgillan, 2016-08-31:
Rob <- annoyed he didn't look like he usually did.

I was flipped here by their support process, so you know what they say about assuming things.

Need to not let my annoyance with re-branding people cloud my actions. Sorry once again and thanks so much for Cura.  Our startup relies upon it (justifiably so)

r