daid/LegacyCura #1355: Feature request: show current layer / ETA while printing?

geeksam, 2015-10-29:
I started using my Printrbot with Repetier, and my daughter (age 7) is really into asking "how many layers are there?" or "what layer is it on now?"  Since switching to Cura, I've had to explain to her that I can't answer those questions any more, which is a little disappointing for both of us.

It would also be nice to have an estimate (even if it's wrong) of about how long is left, so I can make choices like "do I have time to eat before I can play with this print?" or "should I go to bed now or wait until it's done?"

Neither is hugely important, but since I'm here already, I thought I'd mention it.  :)

iashwinprasada, 2016-01-02:
i was working on this! ill be happy to submit a pull request if the mods agree :) 
### *Len|mm* - length of filament extruded
### *L:00000* - Current Layer
### *TL:00000* - Total Layers
[image]

BagelOrb, 2016-01-04:
Print time estimation is performed on the firmware for Ultimaker printers. I don't see any incentive to implement these things for Ultimaker.

Anybody is free to submit a pull request with the code changes necessary to implement this, though.

closed 2016-01-04 by BagelOrb