daid/LegacyCura #1361: Wrong spacing calculation between objects on same table printed one by one and if option "Platform adhesion type" = "Raft" is on

dany-x, 2015-11-09:
Printed with printer "UltiMaker 2" and the inbuilt Profile for the "UltiMaker 2". The printhead crashed into the first printed object while printing the "raft" for the second object.

nallath, 2015-11-09:
Which version did you use?

BagelOrb, 2015-11-09:
The bounding box shadows to keep objects apart didn't take the raft into account. In the upcoming release (Cura 15.10) some related problems are solved where the raft could fall partly outside of the build plate.

However, for one-at-a-time mode the problem still persists.

This is a problem both in the old Cura (version <= 15.04) and the new Cura (version >= 15.06, see https://github.com/Ultimaker/Cura)

dany-x, 2015-11-11:
Hello,

I use Cura version 15.04.3.

Finally we did increase  the value "Head size towards Y min (mm)" under 
menu "Maschine settings..." from its original 10 mm to new 12 mm. With 
this little adjustment the print head will not crash anymore into 
already existing printed model on same platform.

We wondered why all with template "UltiMaker2" defined values for 
"Printer head size" are smooth numbers like 40; 10; 60; 30; But if you 
measure the real values (a hard task with print head mounted to the 
printer) the values are not smooth numbers They are more likely 
something like this 42; 12; etc. The real values (plus 2 or 3 mm) should 
be taken into account in the maschin's template instead of smooth and 
most of all dangerous underestimated values.

Anyway the crash which occurred with our printer with value 10 mm 
instead of 12 for Y min. was not a full crash. Nevertheless did the 
small vertical sheet metal on the left hand front side of the print 
head, touch and hang onto the left vertical edge of a cube which was 
printed already on the same platform - during a x-axis move from left to 
right and while printing the outline of the raft for the second model on 
same platform.

With kind regards
Daniel

BagelOrb, 2015-11-11:
I've changed the machine defaults for the new Cura (version 2.1). Now the machine_head_with_fans_polygon is expanded by 2mm in every direction.
Thanks for taking these measurements to give better estimates!

The raft still causes problems with one-at-a-time printing.

BagelOrb, 2016-03-07:
This should be fixed in Cura 2.1

closed 2016-03-07 by BagelOrb

daid, 2016-03-08:
Note this is also fixed for the 15.04.5 maintenaince release.