daid/LegacyCura #1363: Dragging lines

joebousquet4, 2015-11-16:
While printing this object, Cura adds shell where it's not supposed. This is not part of the STL.

[image]
[image]
[image]

BagelOrb, 2015-11-16:
I believe this is part of the STL. Can you supply the STL?

Also you should take a look at the STL in x-ray view; there might be something wrong with your model (highlighted in red).

joebousquet4, 2015-11-16:
Sorry but I don't know how to supply the STL. I can't attach it to this thread. But you can see in the image below that it's not from the STL. In fact, I tried another slicing software and it was OK.

[image]
[image]

joebousquet4, 2015-11-16:
Here's a link to the STL...
https://drive.google.com/file/d/0B87vALBg4xs4T1JkZy10TXA1Z2s/view?usp=sharing

BagelOrb, 2015-11-16:
Ah now I see what you meant. I thought you were talking about the top of the letters, but you are talking about the lower middle part of the side...

joebousquet4, 2015-11-16:
Correct!

BagelOrb, 2015-11-16:
This is not a problem in the new Cura anymore.
Cura 2.0 is soon to be released.

Cura 15.04 is in low maintenance mode.

joebousquet4, 2015-11-16:
Thank you. Soon in days or soon in weeks?

BagelOrb, 2015-11-16:
Yes.
:P