daid/LegacyCura #1365: STL where 1 unit = 1 inch is 25.4x too big in Cura. What?

RobertBaruch, 2015-11-19:
When you can't control what scale your 3D program is gonna export, you know it's always going to be the wrong scale for your slicer. Example: Fusion 360 exports to STL with 1 unit = 1 inch. When imported into Cura, it turns out to be 25.4x too big. That's weird. I thought Cura assumed 1 unit = 1mm. Then my model, which is 0.625 inches, should be 0.625 mm in Cura and would need to be scaled up.

Instead, it turns out I have to scale the model *down* by a factor of 25.4. What's wrong?

I'd attach a sample STL file, but github won't let me.

BagelOrb, 2015-11-19:
I guess fusion 360 export in 100ths of inches..
You can attach the STL by changing the extension to .STL.txt

daid, 2015-11-19:
Actually, Cura scales up models by 1000x when it finds they are really small. This to fix models exported in meters. (And, else you could not click on them as they where too small)

STL is semi-standard to 1unit=1mm. I highly recommend you complain at the Fusion 360 supplier to add an option to export to mm.

To prevent UI clutter, we're not adding options to support conversion from inches.

closed 2015-11-19 by daid

nallath, 2015-11-20:
The new Cura does have an awesome plugin for this; https://github.com/nallath/BarbarianPlugin

JesseHamburger, 2016-08-28:
> To prevent UI clutter, we're not adding options to support conversion from inches[...] I highly recommend you complain at the Fusion 360 supplier to add an option to export to mm.

This response is unsatisfactory. While it may be true that Autodesk should include export options in their Fusion360 package, you've essentially just said, "units is every other software manufacturer's problem" and in doing so, implied "this is our user's problem, not ours as software engineers."

Cura should have this option, even if it's buried in a menu somewhere, especially if it's going to do something stupid like randomly scale by 1000x if a part is very small. Either the option should be added or you should figure out a way detect it automatically.

JesseHamburger, 2016-08-29:
Thank you, I saw your comment. This shouldn't be a plugin. It should be an integrated feature.

Dim3nsioneer, 2016-08-29:
I'm designing and exporting models in mm with Fusion 360. I also see no problem if Cura supports mainly the units which are used in 99% of all countries worldwide. btw: Plugins in Cura are not second choice structures as a lot of things in the new Cura are implemented as plugins without the user noticing that it is a plugin.

JesseHamburger, 2016-08-29:
> Plugins in Cura are not second choice structures as a lot of things in the new Cura are implemented as plugins without the user noticing that it is a plugin.

It doesn't really matter whether the code is structured as a plugin or not; the point is that Cura should come with this feature ready to use.

> I'm designing and exporting models in mm with Fusion 360. I also see no problem if Cura supports mainly the units which are used in 99% of all countries worldwide.

The percentage of countries that use one system of measurement or another is irrelevant; a large number of parts, manufacturers, and other organizations use units other than millimeters. Just because it is the standard for 3D printing doesn't mean we can ignore the fact that it isn't the standard in every other case. Sometimes a person might have to model something in kilometers or in inches or in micrometers or in feet if they have a set of plans that they need to interface with.

As one example, perhaps I am trying to create new a part to interface for a tool manufactured in the United States for which I am given plans written in inches. Given that all of the dimensions for the part are in fractional inches, it would be insane to try to model my new part in millimeters, converting each of the several dozen dimensions (complete with rounding errors). I should be able to model my part in inches, export the STL (a unitless filetype, which for some stupid reason is the standard in 3D printing), and open it in Cura in the units that I modeled it in without any problems or extra plugin installations.

nallath, 2016-08-30:
STL will already have the rounding issues, as STL doesn't work with fractions in the first place.

daid, 2016-08-30:
Sorry, this has nothing to do with how you design your parts, or how you make your drawings. It's only the export format that needs to be in mm. Your design can be in inches. Your drawings can be in inches. Your step file for your injection mold can be in inches.

Example, I have a file that I designed and drawn in mm. But I can export it in inches just fine:
[image]
Which is the inverse of what you are doing, but shows the same process. It doesn't matter what unit system we used to design it in. We export it in the proper unit. For STL this is mm, even if there is no standard saying so, everyone sort of agreed on this convention.

Every other CAD package that I've worked with (and I've used quite a few) allows for this setting during export. Fusion 360 seems to be the exception here. Not Cura.