daid/LegacyCura #1366: I cannot type into the command line using the keyboard of my laptop?

christytrent, 2015-11-20:


daid, 2015-11-20:
Not enough info. Cura has no commandline.

closed 2015-11-20 by daid

BagelOrb, 2015-11-20:
This seems to me to be a problem of your commands line, your keyboard or your laptop.

christytrent, 2015-11-20:
The keyboard works,  I can type with it for other things in cura such as temp changes. I also plugged an external keyboard into the laptop.that didn't work either. I updated the cura software thinking that would do it. But it did not.

nallath, 2015-11-20:
But what command  do you mean? There is no command line in Cura.

christytrent, 2015-11-20:
When I say command line I'm talking about the box you type commands into such as M500. I can click into the box and see the blinking cursor but it will not type. I used to be able to type. As I said in the above comment the keyboard works everywhere else.

BagelOrb, 2015-11-20:
I think you are using other software than Cura. 
What does it say when you go to help > about... ?

If you *are* using Cura, then where would you find such a box?

christytrent, 2015-11-20:
[image]
This is the window I'm talking about. You type commands into this box such as M500 to save, etc. I can put the cursor into the box but it will not type. I can type temp changes etc. but it will not type into this box any more. It did previously. I'm am using the current version of Cura.

daid, 2015-11-21:
There is a box below it where you can type.