daid/LegacyCura #1367: Flat inside roof over small cavity

JohnDHH, 2015-11-20:
I am making small parts (25x25x3mm) with open spaces on the build surface. The spaces are 2mm high, and are roofed over by a 1mm solid layer. When finished, and viewed from the build surface, the roof surface is open textured with about 50% fill. How do I get 100% fill on this surface

daid, 2015-11-21:
Need pictures of model files to assist here.

rewolff, 2015-11-21:
John, The slicer normally tries to fill in that area for 100%. However as the bridge threads don't have any support, they sag a bit before the next layer is added. As normally the thread-of-plastic is squished against a previous layer, it expands outwards. Without any support the plastic stretches out and forms a thinner-than-the-extrusion-nozzle thread. This results in the "looks like 50% fill". 

I myself am currently in the stage: change as little as possible: "i'm getting working prints!". So I'm printing at a 0.105 mm layer height. This results in "too little plastic" to even form the bridges. It takes a bunch of layers to actually fill the bridged area. 

Is there a setting that allows me to control the layer-height when making a bridge? I think the layer height should be about 0.2mm or even more to get better results than what I'm seeing.

JohnDHH, 2015-11-24:
Rewolf, Would a triangular, or Semi-Circular roof fill more solidly?

rewolff, 2015-11-24:
Yes, up to a certain angle, "overhangs" can be printed perfectly. Some cite 45 degrees, others even further: 60 degrees. A semi-circle will have a flat part near the top with the same problems (but possibly smaller) as you're describing. The overhangs could create headaches. Then they are called hangovers. Especially in ABS, the piece sticking out will curl up and the printing head will bump into it when crossing over to the other side.