daid/LegacyCura #1368: Not able to package Cura for Windows! No solved resources online..

iashwinprasada, 2015-11-26:
@daid @nallath @hg42 

i'm able to build Cura for debian with no errors or absolutly no errors, but im getting a lot of errors and deadends while trying to build package for windows....

i've built debian packages on ubuntu with ease, im trying to build windows packages from within windows 8 with git bash...

im getting weird errors like OUTDIR , no directory found. [Here](https://ultimaker.com/en/community/16669-trying-to-build-cura-with-packagesh)
i managed to manually copy python-portable files into target dir  and comment the package.sh
but im getting make errors...

i've already googled alot on this, there seems to be no documentation for packaging for windows...can we do something about this? can we get this issue closed ? should we come up with a valid procedure? for the better? 

please respond, thanks

just taging related issues
#498 #823 #1215

nallath, 2015-11-26:
Well this is the legacy version of Cura, so don't expect a lot of updates from our end. The build scripts for the new Cura are a lot better / stable, so I'd advise using those.

Ghostkeeper, 2015-11-27:
Perhaps you have to run the build with administrator rights? Id est run git bash with administrator rights so it can install the resulting binaries to Program Files.

daid, 2015-11-27:
The whole "legacy cura" packaging script is a "works for me" thing. Packaging cura is a complex and error prone process. Most likely you are just trying to re-brand/repackage Cura for your own printer, without contributing to the software. So there is little incentive to help you here.

iashwinprasada, 2015-11-27:
i understand, im trying to get it working using the new scripts...ill
update on that...and about the contributions, we've been contributing to
the marlin firmware with a couple of pull requests pending...we really need
to add custom features to the host to get the marlin features to work in
tandem...lil help would be awesome! thanks for making time to reply

BagelOrb, 2015-11-27:
What features? I suppose you would want the new Cura to work with them in
tandem... Take a look at Ultimaker/Cura

fuding, 2015-12-12:
Replace 7z version

iashwinprasada, 2015-12-14:
thanks for helping out! finally completed building it! can help solve other ppls bugs, let me know :)

closed 2015-12-14 by iashwinprasada