daid/LegacyCura #1369: unfilled space in some layers

hombreolo, 2015-11-30:
Hello
My name is Manolo and I'm from Barcelona
I have a problem with this piece
the design is good, the piece looks good but when I see no cap to cap a blank in the last layers
when printing the part it is true, left blank and not filled
you can give me a solution?
Thank you
[image]
[image]

BagelOrb, 2015-11-30:
Can you provide a screenshot of the X-ray view?
If there are red areas it means your design is not good.

hombreolo, 2015-11-30:
yes
is good
[image]

nallath, 2015-12-01:
It might be the poor compression, but I think I see some red parts.