daid/LegacyCura #1371: Stringing because of a z-ramp instead a straight downward movemnent

Martinello, 2015-12-05:


BagelOrb, 2015-12-05:
Is this a bug in wire printing?

Can you explain better what the problem is exactly? Perhaps a picture can
help a lot.

BagelOrb, 2015-12-05:
Never mind my last comment

Martinello, 2015-12-05:
I'm new to Cura and I think there is no way to to route it manually.

The head should move straight upwards, then travel 90 degrees to the next position and the 90 degrees straight downwards like it does when it moves from the left building block to the right one.

Generating a ramp is not what I understand about z-hopping, it starts stringing as it wipes the filament off when it moves over the edge of the left building block.

Martinello, 2015-12-05:
Looked at the code.

Cura sets for the travel from right to left a Z value, so a ramp is generated, resulting in stringing:

[image]

Deleting the Z value in line 120 makes everything fine (+197 other Z values in the code):

[image]

For the travel from left to right, Cura sets no Z value, no ramp, no stringing:

[image]

I don't understand why Cura sets a Z value for the end point of the travel from right to left.

For this small test building block I have to delete manually 198 Z values, not very funny ...

Or can I switch off this behaviour somewhere in Cura?

Ghostkeeper, 2015-12-05:
It looks like a bug indeed then. Cura has come a lot further in the meanwhile though. Perhaps you could try [the version 2 beta](http://software.ultimaker.com/Cura_closed_beta/15.10/Cura-1.99.0-dev.20151117-Darwin.dmg)?

Martinello, 2015-12-06:
Solved my problem, thanks Ghostkeeper!

closed 2015-12-06 by Martinello

BagelOrb, 2015-12-06:
How did you solve it?

Martinello, 2015-12-06:
I'm using Cura 2 beta (1.99.0-dev.20151117) now.
It sets the z-hop in 90° angles, no ramps anymore.

I can't see the z-hop in Cura 2 beta even if set very large values (display problem?) but in Repetier I can see them and the printer is moving so as well.

BagelOrb, 2015-12-06:
OK thanks for the info. Indeed z hops aren't displayed in the new Cura. I
don't know if they ever will be.