daid/LegacyCura #1373: Slicing freeze

warmwhisky, 2015-12-09:
I am continually having my models freeze either while loading or while slicing.

There seems to be no rhyme or reason to it. If I repeatedly close Cura and then reopen the same model it will eventually slice and I can then print. If I open a model of a box (8points) it handles fine, but I want to print things with a little more detail. I have successfully imported 300mb obj files. I cant use Cura once it is in, only save the gcode if I am lucky. But on an obj file thats 20mb? 

I have stopped making any adjustments to settings after I open a model as this just confuses the hell of out Cura because of that silly setting that makes it auto compute, even if you breathe on it! I am tired of spending more time getting things to slice than actually modeling. There seems to be less and less artistic freedom because it is mostly spent closing and opening cura.

daid, 2015-12-09:
Which version are you using?

warmwhisky, 2015-12-09:
1.99.0-dev 20151101

I have 5 versions installed on my PC from 15.02.01 up to current version. This has always been my problem. There is just no pattern to it happening. Sometimes is hangs when loading. Sometimes it hangs at 30%, 65%, in fact its always different.

BagelOrb, 2015-12-09:
What's your OS?

warmwhisky, 2015-12-09:
Windows 7 Ultimate 64bit
i5 4950 processor
32 gig ram