daid/LegacyCura #1374: wrong mm^3 flow calculation on first layer speed

Tiziano-Berti, 2015-12-09:
The tooltip note on "advanced settings>first layer speed" make a wrong calculation because it does not take in account "first layer line width". The actual calculation is: << FirstLayerHeight x NozzleSize x Speed = mm^3 Flow >> but should be: << FirstLayerHeight x NozzleSize x LINEWIDTH(decimal) x Speed = mm^3 Flow >>

BagelOrb, 2015-12-09:
Which version of Cura are you talking about?

Is this about 15.04?

Tiziano-Berti, 2015-12-09:
Yes, last one of the old framework 15.04.3, and precedent

BagelOrb, 2015-12-09:
Well... That doesn't look like a big bug to me.

You should take a look at the new Cura (https://github.com/Ultimaker/Cura)
We're not really developing on the old Cura any more.

Tiziano-Berti, 2015-12-10:
yes not a big issue, just useful.
I'm still using the old version for easier machine setting of repraps, deltas and so on, but i've not see the latest version so i'll have a look soon.
Btw the new framework seems a lot better for customisation, just a bit harder for the moment (i'm not so good in editing code of the json files), but i hope will became easier in the future.
Thank you for fast replies, and for your work on this really good soft :)