daid/LegacyCura #1378: 2 new functions needed and 1 problem

Donaldus57, 2015-12-10:
Hello Cura Team ,

2 new functions are really needed

1.  a seperate measure function  to measure from point to point , somting e.g. i have to make a item smaller and need to know if correct from hole to hole .

2.  when placing struts or gitter  it would be nice to have a direct preview  so that on knows if it is okay before saving ti sd card

also  when reading a gcode from the sd card  the program has problems showing the item !

Best regards from Germany

Don

Ghostkeeper, 2015-12-11:
Hi Don.

1. This feature is expressly omitted from Cura. Cura is designed as a program to prepare a model for printing, not as a program to design or edit models. There are exceptions, for scaling and mirroring stuff, but editing a design in the way that you describe is out of scope. Most 3D modelling software will have the feature you request though. And if you just want to measure things, I would recommend CloudCompare.

2. Placing support manually is not currently implemented. There is a preview function though: The layer view shows exactly the path that the resulting g-code will take, once slicing is completed.

3. That's a bug there! I propose that the rest of this issue will be about this bug. I'll change the title accordingly.

Donaldus57, 2015-12-11:
also  another problem

sometimes a item must be turned for fitting to the table for printing

Problem is  the settings are at 15 degrees  and once in a while i need a 5 degree setting !! or less !!

BagelOrb, 2015-12-12:
Please start separate issues for each problem.

Rotating by smaller steps can be performed by holding the shift or alt key
( I forgot) while rotating.

BagelOrb, 2015-12-12:
Also: what problems are you having when showing gcode from file?

Note that this function hasn't been implemented in Cura 2.0 :(

Donaldus57, 2015-12-12:
Hello Tim,

most of the time I start Cura , must load a Gcode , see it loading, and 
then nothing shown on the Table  upper right side i press the view 
button try to change  ......... nothing shown

Best Regards from Germany

Don Burkholder
Am Bismarckplatz 4 b
48455 Bad Bentheim, Germany
Tel. +49 5922 9827999

This e-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error)
please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail
is strictly forbidden. World Wide Internet Fraud emails about Geld Laundry, Inheritance, u.s.w will be directed to the FBI, EFCC Nigeria, Interpol and British Secret Service.

Donaldus57, 2015-12-12:
Hello again Tim,

also what i need in cura is a measuring tool !

e.g.  a Servo holer for a 450 trex helicopter

I printed one for a Friend and today he came to me and asked me if i can 
enlarge it  because the one i printed only had a room from 31mm  and he 
needs 33mm

okay i can enlarge it  but no way to measure if the room is large 
enough  before printing it :-)

I think this tool would be great for many cura users

Best Regards from Germany

Don Burkholder
Am Bismarckplatz 4 b
48455 Bad Bentheim, Germany
Tel. +49 5922 9827999

This e-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error)
please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail
is strictly forbidden. World Wide Internet Fraud emails about Geld Laundry, Inheritance, u.s.w will be directed to the FBI, EFCC Nigeria, Interpol and British Secret Service.

BagelOrb, 2015-12-12:
Please make a new github issue for the feature request.

As for the viewing problem: are you sure you are in the layer view? When
loading gcode from file you cannot view the model since the gcode doesn't
have that information. The normal view is empty in that case.

Donaldus57, 2015-12-12:
Hello Tim,

Done  :-)

Best Regards from Germany

Don Burkholder
Am Bismarckplatz 4 b
48455 Bad Bentheim, Germany
Tel. +49 5922 9827999

This e-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error)
please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail
is strictly forbidden. World Wide Internet Fraud emails about Geld Laundry, Inheritance, u.s.w will be directed to the FBI, EFCC Nigeria, Interpol and British Secret Service.