daid/LegacyCura #1379: Measuring tool  for Cura

Donaldus57, 2015-12-12:
Hello Cura Team,

what i also need in Cura is a measuring tool !

e.g.  a Servo holer for a 450 trex helicopter  

I printed one for a Friend and today he came to me and asked me if i can enlarge it  because the one i printed only had a room from 31mm  and he needs 33mm

okay i can enlarge it  but no way to measure if the room is large enough  before printing it :-)

I think this tool would be great for many Cura users

BagelOrb, 2015-12-12:
Such a feature would indeed be nice to have.

Currently we are quite busy with implementing other features which are very important, do it would take a while before we would have such a tool in Cura.

When it would be implemented it should involve a mechanism for finding the vertex closest to the mouse pointer, which is quite hard to implement. 

In the soon to be released cura, there is a way to view the total width, height and depth of a model, which should get you a long way.

Donaldus57, 2015-12-14:
Hello Tim,

Nice , also i would like to know if Cura will work without any problems 
on a Windows 10 Laptop ?
I might be receiving a new Laptop this xmas.

Further , just want to let you know that i love this program !! Easy for 
beginners and works good :-)

Best Regards from Germany

Don Burkholder
Am Bismarckplatz 4 b
48455 Bad Bentheim, Germany
Tel. +49 5922 9827999

This e-mail may contain confidential and/or privileged information. If you are not the intended recipient (or have received this e-mail in error)
please notify the sender immediately and destroy this e-mail. Any unauthorized copying, disclosure or distribution of the material in this e-mail
is strictly forbidden. World Wide Internet Fraud emails about Geld Laundry, Inheritance, u.s.w will be directed to the FBI, EFCC Nigeria, Interpol and British Secret Service.

Ghostkeeper, 2015-12-15:
We test on Windows 10 as well. Should work!

There was one bug about a week ago with Windows 10 and large nozzle sizes. It's currently in testing phase awaiting a Windows build. For the rest we have no open issues regarding anything Windows (except that the build process should be better documented).

nallath, 2015-12-15:
I'm not aware of any issues with windows 10. There could be some issues, but I don't expect major problems there.

daid, 2015-12-15:
Cura is not a CAD tool. Measuring is done in CAD. To prevent UI clutter we're not implementing a measuring tool per default in Cura.

The new 2.0 Cura has plugins, so a measuring tool could be made as a plugin.

closed 2015-12-15 by daid