daid/LegacyCura #1381: Cannot slice this OBJ standing upright

foreachthing, 2015-12-15:
This block [cp_015812_fixed.zip](https://github.com/daid/Cura/files/62200/cp_015812_fixed.zip) I modeled does not slice in Cura (15.04.3) - the preview (normal view) looks good, but is not sliced (layer view) in any way.
The attached OBJ comes from Slic3r .. it's a repaired version of an STL (which doesn't work either in Cura).
You can see, that the hole in the center is not sliced.
Now, rotate the block (green ring, 90°, no support required for the U-shape). Suddenly, the hole and the thread are sliced - wth? Am I missing a setting?

nallath, 2015-12-15:
Did you check it in x-ray view?

daid, 2015-12-15:
Slices fine with default settings. Most likely you have the "type B" mesh fixing checked, which removes internal holes on the 2D slices.

closed 2015-12-15 by daid

foreachthing, 2015-12-15:
Ok ... I removed that option and it slices correctly.
But why would it remove the hole in a different orientation?

daid, 2015-12-15:
Because TypeB applies some very aggressive algorithm in an attempt to fix the model, and thus should be used with care.