daid/LegacyCura #1383: Pronterface UI not responding

glen-cafferty, 2015-12-16:
When the Pronterface UI appears: the temperature boxes accept input, but that's all that responds. The "Connect, Print, and Cancel" boxes are greyed out, none of the motor controls respond nor is it possible to enter anything on the command line. Everything was working until yesterday. I'm printing on a Printer Simple Metal.

BagelOrb, 2015-12-17:
This is the repo for Cura, not Pronterface.

closed 2015-12-17 by BagelOrb

glen-cafferty, 2015-12-17:
I get the same thing when I try to print from the Basic UI. It comes up with "Printing on COM 6" on the top line; "Detecting baudrate" and "Temperature: 0". Buttons are grayed out, and it won't accept any input. When I turn the printer off I get "Error: Serial exception: call to Clear Comm..." and "Temperature: 0". The "Connect" and "Error log" buttons become available. 
Error log reads: 
Baudrate test ok: 4 
Send: M105 
Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:21.4 /0.0 T2:37.5 /0.0 @:0 B@:0 
Baudrate test ok: 5 
Send: M105 
Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:21.4 /0.0 T2:37.5 /0.0 @:0 B@:0 
Baudrate test ok: 6 
Send: M105 
Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:21.4 /0.0 T2:37.5 /0.0 @:0 B@:0 
Baudrate test ok: 7 
Send: M105 
Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:21.4 /0.0 T2:37.5 /0.0 @:0 B@:0 
Baudrate test ok: 8 
Send: M105 
Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:21.4 /0.0 T2:37.5 /0.0 @:0 B@:0 
Baudrate test ok: 9 
Send: M105 
Recv: ok T:0.0 /0.0 B:0.0 /0.0 T0:0.0 /0.0 T1:21.4 /0.0 T2:37.5 /0.0 @:0 B@:0 
Send: M999 
Changing monitoring state from 'Detecting baudrate' to 'Operational' 
Recv: Resend: 1 
Recv: ok 
Recv: ok 
Recv: Error:0 
Recv: : Extruder switched off. MINTEMP triggered ! 
Changing monitoring state from 'Operational' to 'Error: 0: Extruder switched off. MINTEMP t...' 
Recv: Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting) 
Unexpected error while reading serial port: SerialException: 'call to ClearCommError failed' @ machineCom.py:_readline:516 
Changing monitoring state from 'Error: SerialException: 'call to ClearComm...' to 'Error: SerialException: 'call to ClearComm...' 
Connection closed, closing down monitor 

Any help would be appreciated. 

Glen 

----- Original Message -----

This is the repo for Cura, not Pronterface.

KiteLab, 2015-12-17:
```
Recv: Error:0
Recv: : Extruder switched off. MINTEMP triggered ! 
```
Unconnected thermistor for the first extruder - or broken wire.

Ghostkeeper, 2015-12-18:
The guys over at Pronterface would probably like to hear about this: https://github.com/kliment/Printrun/issues

nallath, 2015-12-18:
Your printer is broken. Cura can't fix that.