daid/LegacyCura #1384: Impresions double tete

jbs37, 2015-12-21:
Bonjour a tous, 

Je suis nouveau dans le monde de la 3d. j'ai acheté une Bigfoot, avec double extrusion mais je n'arrive pas a en faire l'usage.

Je voudrais que la deuxième buse serve a imprimer les supports d'impressions mais je n'arrive pas a le programmer dans cura. 

Help me please ^^

Ghostkeeper, 2015-12-21:
L'utilisation d'un deuxieme buse pour les supports n'est pas pris en charge avec cette version. Il est sur notre roadmap, mais je ne peux pas vous dire quand il est fait.

jbs37, 2015-12-21:
Merci Ghostkeeper pour votre rapide retour,

Vous parlez de la version de cura ? si c'est le cas, je viens de la désinstaller car je me suis aperçus qu'il y avait une version beaucoup plus récente... J'espère que ce sera bon ^^

Ghostkeeper, 2015-12-21:
To summarise for English readers: jbs37 asked whether he could print supports with a different nozzle with dual extrusion, which is not possible in this version. I suggest that this thread can be closed (but don't have the rights to do that myself).

daid, 2015-12-21:
[image]
Actually, you can just select it just fine. But dual extrusion in this version of Cura is limited, no configuration for different materials are possible.

closed 2015-12-21 by daid

Ghostkeeper, 2015-12-22:
Didn't know that. Sorry for the misinformation!