daid/LegacyCura #1385: Strange infill behavior under 15% overlap

rrichter90, 2015-12-23:
I'm using version 15.04.3 on Linux.
By decreasing the infill overlap from 15% the infill seems to shift out. This effect disappear at 0% overlap

[image]

daid, 2015-12-23:
What happens if you add more shell?

BagelOrb, 2016-01-04:
Can you please supply the model which you sliced?
We can then try to reproduce the problem and see where it goes wrong.

rrichter90, 2016-01-04:
Sorry for my late response. It also happens with more shells. 
You can download it here: https://www.dropbox.com/s/0u02xo1w87su07w/model.stl?dl=0

BagelOrb, 2016-01-06:
I cannot reproduce it in the newest version of Cura. Soon the newest version will be released officially.