daid/LegacyCura #1386: Will not print raft with sphere with hole in middle

iseries1, 2015-12-28:
Cura will not lay the object flat when the object is a sphere with a hole in the middle.

iseries1, 2015-12-28:
Sorry, wrong words.  If the object is a cone with a hole in it.  Object is above the base and will not lower nor will the raft print up to touch it.

nallath, 2015-12-29:
The lay flat algorithm is rather naive and gives no guarantee that the found solution is the best one (or correct at all)

iseries1, 2015-12-29:
Since you can move the object in all directions except the Z axis it is rather difficult to position the piece on the bed so that you can print it.  For example if you scale the piece it becomes unprintable.

BagelOrb, 2016-01-04:
This issue is about the drop on platform feature, not the lay flat feature..
What version of Cura did you use?

iseries1, 2016-01-05:
Sorry for the late reply.  Cura 15.04.3
The issue: Take a cone with the point side down.  Cut 3 or 4 mm off the bottom so it is not touching the base surface.  Extract a STL file and load it into Cura.
Select Raft with .1mm layer height and view the layers.  Layers 1-4 will be empty until layer 5 where it will start to print the part.
I worked around this issue buy using the Export Settings option under Expert and set the object height lower by a couple of tenths of a mm.
This also happens if the object is scaled smaller.

BagelOrb, 2016-01-06:
Ah yes, probably the first couple of layers were too small to be printed and therefore left empty.

This problem has been solved in the new Cura. All first empty layers are discarded and consecutive layers are moved down.