daid/LegacyCura #1387: Cura 15.04.3 Mac and Linux - Holes on the top layer

Almisuifre, 2016-01-04:
Hello,

I'll allow me to question you on a problem encountered on Ultimaker² Prusa and sliced ​​with Cura.

I found that I had holes in the top layer of my printed pieces. The filling is only little on the result.

It seems that the filling calculation (by 20%) + speed (here 50 mm / s) + distance is not included. Extrusion is very slow in completing. The amount of material is small.

Would it be possible to make an update including a correct calculation + so: Filling (20%) + speed (50mm / s) + distance + (extrusion adjustment value) can offset this problem and some either filling or speed.

This valeut can be charged in the first panel with the filling ...

Thus, 100% of filling the adjustment value is zero.
Therefore 20% + filling this value add an amount equal to the distance or more.

It seems to me that this problem has lasted too long. The solutions that are much data is still risky and brings waste of time + material loss ... Cura is not / more competitive! This is unacceptable...

This is a verry bad result to 50mm / s 20% full.
[image]

Thank you very much for your prompt and future improvement.

BagelOrb, 2016-01-04:
This repository is for Cura 2.0 (and Cura 15.06 and Cura 15.10), not for Cura 15.04, which is the old Cura.

The picture you included shows the problem of pillowing / warts, which is not a problem in the calculation of any value, but a hardware problem. The gaps in the infill need to be crossed by bridging, but hot air trapped in the infill causes the bridges to collapse and curl upward.
To reduce this problem, simply increase the top layer count or the top/bottom thickness.

It is unclear to me what calculation you are referring to and the formula's you use don't make any sense to me. Can you maybe try to explain better what value needs to be calculated differently and in what way?

If this problem has been bugging you for a long while you should have asked for help sooner.

Do other slicers solve this problem in a way which Cura doesn't?

Almisuifre, 2016-01-04:
Ok ... Let for version.

Before everything was perfect. The problem occurred in this version and more. I have also found it about two years ago and then, well then.

Well, let's take the math. In absolute terms, on the bed, if 1cm step ahead, we will 1cm extruded right?
What actuelement in current vesions?

Would it be possible to insert a corrective? This would provide a little more filament such that: a infill 20% when the feed head 1cm, 1.1cm extruded filament may be 1.2cm, 1.3cm spots? Depending on the results desired.

BagelOrb, 2016-01-04:
Are you French? The auto-translation is pretty bad. Perhaps try without translating...

You can change the Flow percentage in the new Cura and on the firmware of Ultimaker printers. Is that what you are looking for?

Almisuifre, 2016-01-04:
C'est exact, Français... Si la traduction est mauvaise, m'en voulez pas... Bref !

Alors, oui, en quelque sorte, c'est le débit que j'aimerais changer. Cependant, je n'ai besoin de modifier ce débit qu'au moment de finir les couches "Top Layer".

En gros, First Layer OK
Remplissage interne (un peu plus de matière)
Remplissage Top Layer (Beaucoup plus de matière)

Ghostkeeper, 2016-01-04:
La seule possibilité que je vois est l'augmenter la valeur de "Infill Overlap". Cela va augmenter l'épaisseur des lignes de remplissage interne un peu, mais il ne sera pas beaucoup d'aide.

Les nouvelles versions de Cura a une option de changer l'épaisseur des lignes de remplissage interne directement, qu'est ce que vous cherchez, je pense. Mais cette option n'est pas compris dans Cura 15.04.3.

Almisuifre, 2016-01-05:
Ok, je comprends... C'est déjà bien de voir que le souci est pris en compte.

Et pourquoi pas, si s'est possible, j'oblige à rien, de faire comme pour "infill overloap" mais en "top overloap" ? Cela simplifierais car celà gèrerais le pourcentage de chevauchement du passage de la tête.

100%, la buse passe deux fois au même endroit.
50%, la buse repasse à la moitier de son précédent passage
0%, comme actuellement, la buse passe mais à la largeur de la buse...

L'image étant plus parlant qu'un long discours....

[image]

BagelOrb, 2016-01-06:
You misunderstood what infill overlap usually means. Normally that concept is used to signify the overlap between infill lines and the outer walls. 

However, your ideas about top overlap do make sense. In a way it's already implemented in the new version of Cura: you can specify the line width of the top to be smaller than the nozzle width. Then the nozzle will overlap a previous line for every new line you print. This way you have overlap of the nozzle, without extruding too much.

It's generally a bad idea to extrude too much on purpose. While it can solve one problem in one area, it will introduce over-extrusion problems in various other areas.

daid, 2016-01-06:
You have warts. Quickest fix is just increase the top/bottom thickness.

Almisuifre, 2016-01-07:
Ok, that's what I did. Overlap increase the internal walls and the layer of TOP 0.8 to 1mm.
Thank you for this case study.
--

Ok, c'est ce que j'ai fait. Augmenter l'Overloap des murs internes et la couche du TOP 0.8 à 1mm.
Merci d'avoir étudier ce cas.