daid/LegacyCura #1388: Cura not slicing when in machine settings "Machine Center 0,0" is selected

ihouser, 2016-01-05:
Recently I'm updated ArchLinux and Cura stoped slicing. Cura not updated, nothig to downgrade.

Cura not slicing when in machine settings "Machine Center 0,0" is selected. After deselect slicing OK
That is problem fo mine Wanhao Duplicator.

Installed cura-old (from aur cura_15.04.2-debian_amd64.deb)
Linux 4.3.3-2-ARCH #1 SMP PREEMPT Wed Dec 23 20:09:18 CET 2015 x86_64 GNU/Linux

Also I'm tryed to install cura 15.06 – no success: libarcus-git "cmake: error while loading shared libraries: libjsoncpp.so.1: cannot open shared object file: No such file or directory". Installation aborting.

jamescarpino, 2016-06-16:
Same experience.  Running Ubuntu 16.04 and up to date.  Cura 15.04.4.  Slicer doesn't activate, no progress bar, no gcode to save. 
In contrast, Cura engine runs properly in RepetierHost v1.60, with "Cura_SteamEngine version 15.05", configured for a printer with Machine Center 0,0.

BagelOrb, 2016-06-16:
Try cura 2! Download it from software.Ultimaker.Com