daid/LegacyCura #1390: Instructions of bed leveling wizard get cut off.

a4jp-com, 2016-01-07:
[image]

Windows 10
Cura 15.04.3

a4jp-com, 2016-01-07:
[image]

a4jp-com, 2016-01-11:
Any chances of getting this fixed? It might be easy if the window is just made a little bit bigger.

nallath, 2016-01-11:
Very little; We no longer actively do bug fixes on this version of Cura.

a4jp-com, 2016-01-11:
Okay. Is there a new version? ^^

a4jp-com, 2016-01-11:
I can see Version: 15.04.4 Release date: 1/5/16 is the newest version but where would you like me to send info? Or do you mean you have stopped the whole project?

BagelOrb, 2016-01-11:
Yes there is. See https://github.com/Ultimaker/Cura
For beta builds, see http://software.ultimaker.com/Cura_closed_beta/15.10/

a4jp-com, 2016-01-11:
Nice to see there will be a newer version soon or sometime ^^ 

The link http://software.ultimaker.com/Cura_closed_beta/15.10/ seems dead. 

Where can I download a newer copy of the program?

Ghostkeeper, 2016-01-12:
No public builds are being made at the moment since we added things to Cura related to upcoming Ultimaker projects, and they need to be kept secret. I've tried to look after making a new build for the public beta but no luck yet.

a4jp-com, 2016-01-12:
Okay. Sorry to hear that. I hope those modular projects can be removed so the software can be tested and used sometime soon.