daid/LegacyCura #1392: Cura Program Not Displaying Properly

MicheliDE, 2016-01-10:
Hi,

I have an older computer and I had a previous installation of Video driver and Cura that was able to display the new cura interface with the slide menus on the right hand side. When my computer crashed and I reinstalled all of my applications I am stuck with the old cura interface. Can anyone help me resolve this issue?

daid, 2016-01-11:
Try the latest from here:
http://software.ultimaker.com/Cura_closed_beta/15.10/

gotymog, 2016-10-19:
I´m having problems in windows 8.1, could it be my actual version of Python? My issue basically is, the object doesn´t move when I turn it around, it´s like it got stuck until I move the property´s window so it refresh the current position of my stl object.

BagelOrb, 2016-10-19:
Try Cura 2.3 it has been released today!

Ghostkeeper, 2016-10-20:
No such problem on Windows 8.1 here, gotymog. And Cura 15.04 supplies its own version of Python with the installation, so any other installations you may have should not have effect.

If it's not updating the 3D window correctly, perhaps you could try updating your graphics drivers?