daid/LegacyCura #1395: (Suggestion) Print speed percentages

rms95, 2016-01-13:
Hi,

I recently started 3D printing on my own printer, and I noticed something that could be improved a lot.

I'm using 15.04.3, in expert mode.
The general print speed can be configured in the basic tab, and the specific values can be set in the advanced tab, which does not make a lot of sense to me. It would make a lot more sense if the expert tab would provide the option to make the print speed N times the "general speed" (from the basic tab).
This can be done entirely in the GUI of course, the most optimal way I would see this feature is the following (where brackets are input fields):
Travel Speed (mm/s)   [ 1.50% ]  300 mm/s
Bottom .....................   [ 0.80% ]  160 mm/s
.....
Updating the last value when the percentage is changed.

In my opinion this makes a lot more sense, because when I print certain delicate parts, I want to slow everything down or up at approximately the same rate.

Please let me know your thoughts on this topic. :)

daid, 2016-01-13:
We got from there, trust me, it's no better at all, it's much worse.

closed 2016-01-13 by daid

nallath, 2016-01-13:
If you really want this, you could modify this in the new Cura yourself (and see how well that works)