daid/LegacyCura #1398: Missing 'holes' in layer view

qw01er, 2016-01-18:
Hello there. 
I am definitely new to 3D printing and somewhat to 3D drawing. 
My goal is to build a cellphone case with a hole of a certain shape in its back. 
I created this object with Freecad (importing a STL file, creating a solid from it and 'cutting' holes in it) and exported to STL for Cura. 
When I import it in Cura, all the views show the 'holes' except the layer view. Of course, it will print like the layer view. 
I wonder if there is a way for me to upload the STL here, but I don't see a way to do it. 
Any Idea?

daid, 2016-01-19:
Could be that the "mesh fixing" options are causing problems for you. In the expert menu, disable the "fix horrible" options.

qw01er, 2016-01-19:
After playing for a while, I found out that the issue was coming from Freecad. I am not 100% sure of what I did, but after proceeding step-by-step I noticed that transforming to solid the imported STL file may have done the trick. It was not a Cura issue (even if I still don't understand why it is showing in every view except in the layer view). 

Anyhow, this is solved. Thanks for your time anyway.