daid/LegacyCura #1401: Main process ignores slicer errors and fails because result is unavailable

Feuermurmel, 2016-01-20:
While debugging another issue, I ran into this issue:

```
(venv)michi ~/.../cura/Cura$ pythonw -m Cura.cura -s -o test/foo.gcode test/foo.stl
load preferences from /Users/michi/Library/Application Support/Cura/dev/preferences.ini
Failed to listen on port: 49674
Failed to listen on port: 49675
Listening for engine communications on 49676
Traceback (most recent call last):
  File "Cura/util/sliceEngine.py", line 391, in _runEngine
    self._process = self._runEngineProcess(commandList)
  File "Cura/util/sliceEngine.py", line 620, in _runEngineProcess
    return subprocess.Popen(cmdList, stdin=subprocess.PIPE, stdout=subprocess.PIPE, stderr=subprocess.PIPE, **kwargs)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 13] Permission denied
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/Users/michi/Desktop/cura/Cura/Cura/cura.py", line 89, in <module>
    main()
  File "/Users/michi/Desktop/cura/Cura/Cura/cura.py", line 75, in main
    gcode = engine.getResult().getGCode()
AttributeError: 'NoneType' object has no attribute 'getGCode'
```

The underlying problem is that the slicer process tries to read/write from/to a different path than was specified on the command line. Thus it returns a stack trace to the parent process and `_result` is never set in the `Engine` instance which leads to the parent process also crashing.

I'd like to help improving the code. One place that bugs me is the following piece from `_runEngine()` in `sliceEngine.py`:

```py
self._modelData = engineModelData
try:
	self._process = self._runEngineProcess(commandList)
except OSError:
	traceback.print_exc()
	return
```

Why is that exception printed and not passed to the caller instead? I'd like to change this so that I can catch the exception in `main()` and instead print a nice error message. What do you think?

nallath, 2016-01-20:
I'd recommend not putting a lot of time or effort into this version of Cura. We don't support it anymore, so there isn't going to be a lot of active development on this codebase anymore.

Feuermurmel, 2016-01-20:
@nallath Yesterday, I've been told that the old version is still being supported: https://github.com/Ultimaker/Cura/issues/600#issuecomment-172819627

I will have to look into the new version in more detail, but some features I require are still missing. I'm going to report those, if they haven't already, but in the meantime I'll continue with the old version and I'd like to fix the problems that bug me the most.

Ghostkeeper, 2016-01-20:
As far as I understood it's still officially supported, with Ultimaker publishing 15.04 as its recommended version and all, but only Daid is allotted time to develop and he is allotted very little time too, for Cura.

daid, 2016-01-20:
We still support the old version fine. But commandline slicing is something that breaks quite often, as it's not an official feature, but a contributed feature.