daid/LegacyCura #1405: G code error for Lulzbot Taz 5

smithtec81, 2016-01-22:
I'm extremely happy with the way Cura runs the Ultimaker. In side by side tests with MatterControl it is substantially better. However, when I post a program to run the Taz 5 from an SD card, there appears to be a glitch in the code that turns off the nozzle heater after a couple minutes. I would come back to air printing and grinding the filament, so I watched it closely and noticed a minute or so into the print the nozzle temp starts dropping. Lacking a terminal to actually look at the G code, I'm not exactly sure what the issue is, but it is definitely something Cura is doing.

nallath, 2016-01-25:
Well, the lulzbot has their own fork of Cura which changed some things. We (at Ultimaker) don't have a Taz, so we cant really help you debug this. I'd suggest contacting aleph objects about this.

closed 2016-01-25 by nallath

smithtec81, 2016-01-25:
When I click the Help menu in Cura, and select "report a problem", it brings me here.

daid, 2016-01-25:
>I'm not exactly sure what the issue is, but it is definitely something Cura is doing.
Most likely. No. As our Cura does not touch temperatures at all, unless you use specific plugins.
>Cura has Lulzbot Taz available in the machine selection area, so you would think it's been tested to run that machine.
It's contributed, we have no TAZ machines to test on.

Most likely (as 80% of the time with temperature issue), you have a lose wire somewhere.

You're asking us to diagnose a problem with almost zero information, and you haven't even looked into the GCode yourself.

smithtec81, 2016-01-25:
MatterControl runs the printer just fine without temp problems. Ergo, it's not a loose wire. I'm not asking you to do anything, I'm just reporting that I found a problem.
I found Cura 18.03 on the Lulzbot site which says it is Taz specific(perhaps the fork previously mentioned?), so I'll give that a try. No, I haven't looked at the G-code, but I would love it if someone can tell me HOW to look at the G-code...

daid, 2016-01-25:
http://lmgtfy.com/?q=how+to+look+at+gcode

daid, 2016-01-25:
Your success with matercontrol could be down to random chance.

smithtec81, 2016-01-25:
Since MatterControl prints successfully 100% of the time, and Cura loses temp 100% of the time, I'm disinclined to call that random chance. Thanks for the help understanding G code, I would have never thought to look that up on the internet. I guess you didn't know that in the real world of CNC machines, where G code came from, you can read it in any text editor. It's specific to 3D printers to save it in *.gcode file extensions. Which would be why it requires a program that can read the file extension and post it to a terminal window in which I can view or edit. But hey, thanks for all your help, I'm sure you have really important things you need to do.

BagelOrb, 2016-01-26:
Indeed we already knew that. You can view the gcode files in any text editor, just like CNC gcode...

smithtec81, 2016-01-27:
Ah, thanks for that. I just assumed because Windows didn't know how to open the file there was something special about the format. It never occurred to me to open directly in Notepad.