daid/LegacyCura #1407: Unable to select more than one model on build plate

pcollingwood, 2016-01-26:
15.04.4
Windows 10
When loading multiple models, unable to select anything but the first model loaded. Click and drag anywhere in the workspace will move the first object. The only way to move focus is to right click, select  "Split object into parts". and the focus changes to a different part/object  Do this until the required part has focus. The problem after that is if you moved the part and need to move another part, as soon as you do the above process the first part is re-positioned. It does, however, seem to hold any size change you make, just not the position in the workspace.

BagelOrb, 2016-01-27:
Cura 15.04 has never really been tested on windows 10. I'm glad to present
a totally new interface, which we are now arduously testing on several
operating systems before it will be released.
Cura 2 will be a lot better.

daid, 2016-01-27:
In general, this has been linked to color settings. I know this happens when you put your system in 16 bit color mode.

pcollingwood, 2016-01-27:
I had seen that as an issue in a different thread, but for me it broke after I installed Win 10. I don't think this has anything to do with my colour mode being too low, not that I can find where to adjust it in win 10 :-(

pcollingwood, 2016-01-28:
Daid, would you like me to play with it on my system?

Cheers
Pete

pcollingwood, 2016-01-29:
Sorry, that previous message was for BagelOrb.....

BagelOrb, 2016-02-01:
If you would play around with it we could inform other users of your
findings. However, a totally new Cura GUI is being released in a month or
so, so there is not much use in playing around with the settings.

pcollingwood, 2016-02-01:
No worries, I'll wait it out then

Regards

Pete