daid/LegacyCura #1410: Monoprice 3d select aka wanhao i3 aka prusa mendel note running end.gcode?

arcadepcnut, 2016-01-28:
Hello. my first printer and my first printer post! so this is definitely user error.

I have printed 3 objects using the default setting prusa mendel i3 gcode. All 3 have printed wonderfully. Hurray. However at the end. my printer stops in the center right above the part. Cura 15.04.4 still has status bar up that says printing on com3
printing 
termperature: 190 Bed:45 and the green status bar is full and only option is "cancel print" I have to cancel print (even though its done!) every time and turn the printer off and move tray out manually to get my part. 

I am afraid since its not running the gcode to do all the stuff the gcode does its going to mess up my printer. How do I fix this? I have waited 30 minutes after my first print thinking this was the cool down feature. However it never finishes cleanly and once I cancel print io have to close program and reload it to do next print

One thing I must tell you is originally I setup the setup wizard to use other then custom then setup before realizing I could use the prusa mendel i3 setting. maybe that's the problem?

Ghostkeeper, 2016-01-29:
I don't think that'd be the problem. Probably a bug! It's hard for us to keep up exactly how each printer works with things like beginning and end g-code, especially since there is no Wanhao i3 to test with here at our office.

arcadepcnut, 2016-01-29:
I would be more than happy to test. I'm on my 10th print and manually have to shutdown printer each time. 

Eric

BagelOrb, 2016-01-29:
If you can look at the end gcode and fix it, then send it to us and we will
include it in the profile...

arcadepcnut, 2016-01-29:
Oh I don't know that much about it. I have only had a printer for 2 days. I can't imagine what going on. The printer just stops right above the part and then the software freezes. I'm only assuming that it's not running the end gcode because it doesn't do anything in that section. Is there a log file I can look at?

Eric

BagelOrb, 2016-01-29:
The only log which would be interesting is the log of the firmware on the printer itself. We at Ultimaker don't know much about your printer; you should contact them instead of us.

closed 2016-01-29 by BagelOrb

arcadepcnut, 2016-01-29:
Ok I'll try a different software. I need something that works. When you are trying to learn "yeah we have a bug. Fix it yourself. Close" is not very helpful. 

Eric

BagelOrb, 2016-01-29:
I'm not saying you should fix it yourself; I'm saying that you should seek help from the manufacturers of your printer. We don't develop the software for your printer. All parts of Cura which are specific to your printer have been included by third party sources, which wanted to add the functionality to Cura. 

We at Ultimaker make software for our printers, but make it so that anyone can contribute to it so that the software is usable for their printers also.

So the bug is really not in our code; you could get the manufacturers of your printer to correct the mistake in the end gcode.

arcadepcnut, 2016-01-29:
Oh ok. I understand now. Sorry. 

Thank you for your help. I tried some other software briefly and I like cura. So I'll work around it. If I come up with anything I will let you know. 

Eric