daid/LegacyCura #1412: Cura goes unresponsive (beachball) on Mac during long print.

kilpatds, 2016-02-08:
Opening new issues because issue 1115 was closed.

It goes unresponsive. The print finishes fine, but the laptop has to spin the fan up as it's taking 100% of a CPU.

If I force-quit it, a 2nd Cura process then spins up to take 100% of CPU. I briefly trussed it, and it was a python process in a tight loop, but I don't have details handy.

Cura version: 15.04.4
Printr: printrbot play

The next time it happens, I'll try and attach some debugging info.

daid, 2016-02-09:
USB printing?

kilpatds, 2016-02-15:
Yep, USB printing.  samples attached.  From the C layer, of course, so I don't have the python stack. I'm hoping the loop context will be obvious from the operations?

Cura-Loop*.txt are from the main program.

[Cura-Loop.txt](https://github.com/daid/Cura/files/130191/Cura-Loop.txt)
[Cura-Loop2.txt](https://github.com/daid/Cura/files/130192/Cura-Loop2.txt)

Cura2 is from some other program that starts after the main program gets killed.  It also get into an infinite loop.

[Cura2-Loop.txt](https://github.com/daid/Cura/files/130190/Cura2-Loop.txt)

To be honest, if Cura just didn't forget that I had already set it up, and remembered base preferences when it's force-Quit, this wouldn't be an issue.

kilpatds, 2016-02-15:
For example, while the top of the loop is python called from xPyCallback::EventThunker(wxEvent&), the loop covers repeated calls to _BitmapFromBufferAlpha, which I assume has a pretty thin python shim, followed by _BitmapFromBufferAlpha, and some calls to _BitmapFromBufferAlpha

IOW, it looks like it's looping on drawing.