daid/LegacyCura #1415: can't package on my macos

MusicJelly, 2016-02-18:
I have a mac-mini with OS X El Capitan (version 10.11.3).
And I bulid my python env follow READE.md,
Now I have a python with 'non-system',  'framework-based', 'universal', 'deployment target set to 10.6', also, I bulid my virtualenv python and install wx, PyOpenGL, numpy, pyserial, power, setuptools, py2app, modulegraph, pyobjc. 
Etc.

Then I can run cura by pythonw in my mac. Very good.
**_But I can't package it_**
I can run the package.sh actually, and it bulid a app finally.
But when I open the app, it arise a window with 'open console' or 'terminate'. Then I run the Cura in <app>/Contents/MacOS/Cura in my terminal, and it tell " 'module' object has no attribute '_objc' ".

I think that maybe my py2app or objc is worng. Then I have upgrade my py2app from 0.7.2 to 0.9 and upgrade modulegraph from 0.11.1 to 0.12.1, but it is still same problem.

Fornow, I have try everything I think. But it is still.
Now my python is 2.7.3 with wx(2.9.4.0), PyOpenGL(3.1.0), numpy(1.10.4), pyserial(3.0.1), power(1.4), setuptools(20.1.1), py2app(0.9), modulegraph(0.12.1), pyobjc(3.0.4).

I don't know where I am wrong, 
**_why I can run cura in my Mac but I  can't package it (Actually can package but not to run app)._**

Can you tell me the version of your python module. Thanks.
Also, I can sent you the app that I package if necessary.

jmduke, 2016-03-04:
+1: actually running into the same issue!

daid, 2016-03-04:
The problem is, we never managed to get the packaging to work properly on newer versions of OSX. Cause even if you target python for 10.6, newer versions of OSX decides that, well pretty much, "fuck you". And it does a few things not for 10.6, breaking everything.

We have 1 machine that manages to build it with success for OSX. Nobody is allowed to touch that machine. And even we are not able to reproduce this result on other machines.

MusicJelly, 2016-03-05:
ok, I see. I will try anything for it. 
Thanks for your help !

donaldding, 2016-03-23:
got the same problem, maybe is the py2app ,@daid l want borrow that machine