daid/LegacyCura #1417: Extraction of PortablePython_2.7.2.1.exe not working in package.sh

diracsbracket, 2016-02-23:
Hi...
I am trying to build Cura in Windows using MinGW.
The extraction of the PortablePython .exe archive, initially seems to work:
`...
Extracting PortablePython_2.7.2.1.exe $_OUTDIR/App
Extracting PortablePython_2.7.2.1.exe $_OUTDIR/Lib/site-packages
...
`
but when looking at the log.txt file, it twice says: 

`No files to process
Everything is Ok
`
The other extracts (pyserial, PyOpenGL, ...) seem to work correctly.

Later in the script, I get:

`mv: cannot stat `$_OUTDIR/App/*': No such file or directory`

I don't know how the _OUTDIR must be defined, it seems that is a variable used internally by 7z?

Why is the above not working?
Thanks!

daid, 2016-08-05:
"$_OUTDIR" is the name in the zip file, it's not a variable.

huanghui, 2016-09-03:
me too, Does anyone solve the problem