daid/LegacyCura #1418: Configuration wizard crops content, unresizable, unscrollable

jonhnet, 2016-02-23:
[image]

First time user, on page three of the modal Configuration Wizard; typing this bug report under the un-dismissable splash screen.

I'm looking at a list of "The following pre-defined machine profiles". The last entry is truncated, suggesting there are more entries below. The window is unresizable, the list is unscrollable.

It would also be good if the entries were links to web pages or a wiki or something. I'm building an MPCNC, and I have no idea which of these names describes a printer most like the thing I'm building.

BagelOrb, 2016-02-24:
Hah! There has been so much user contributed content that it doesn't fit anymore. 

:D 

I guess it's also up to users to fix this problem then...

daid, 2016-02-24:
> Hah! There has been so much user contributed content that it doesn't fit anymore.

on linux.

jonhnet, 2016-02-24:
I think I encountered this problem because my desktop has an unconventional screen scaling. Here I installed Cura in a fresh Ubuntu 15.10 vm. It looked fine at first; when I changed the font scaling factor, it got wonky:

[image]

Note that even the fixed text fields don't fit anymore. I think the layout for this window is weirdly brittle. It would be nicer if the fields expanded to include their contents, and then the outer container provided a scrollbar as needed.

The overall window should be resizable; I could have worked around this problem by growing the window. Why not let me use available real estate?

The window (and the blue splash screen) shouldn't be marked stay-on-top -- that's just being rude to the user. What if the user wants to use a web browser to search for the terms in the dialog box? She has to use whatever pixels aren't being claimed by the modal dialog.

Off to git pull and see if these are easy fixes.

jonhnet, 2016-02-24:
Wow, wxWindows is remarkably resistant to fixing this problem. I tried adding self.SetScrollbar() to InfoPage, and got a scrollbar, but the text laid out inside the scrollbar didn't actually move.

Furthermore, with the font scaling exaggerated, the contents of the window don't actually get longer -- it'l like the text spacing is based on the nominal font size, and then the font layout engine is stuck cramming scaled-up text into a too-small box:
[image]

My enthusiasm for fixing this bug is waning...

jonhnet, 2016-02-24:
Yup, I tried sizer.SetVGap(23), and the sizer dutifully allocated more space, which made wxwindows dutifully allocate a bigger (but unresizable) top-level window. But all of its allocations seem to be based on the wrong font scaling factor.

I think I'll let it rest here, but hopefully these notes will help the next person who wants to fix this accessibility problem. I want to actually start playing with Cura. :v)

daid, 2016-02-25:
You'll most likely have to wrap it all in a scrolled panel:
http://www.wxpython.org/docs/api/wx.lib.scrolledpanel.ScrolledPanel-class.html

BagelOrb, 2016-02-25:
Do note that a totally revised frontend is being developed for the next
Cura release. It would be a shame to put a lot of effort into fixing the
old cura with the new Cura around the corner..