daid/LegacyCura #1422: The empty space is filled

iHateInventNames, 2016-02-29:
[The STL object](https://drive.google.com/file/d/0B7iRuJQSQAVWb25YQ0p2ZksydzA/view?usp=sharing) slice result gives no emptinnes in resulting gcode.
The Cura itself shows no the problem in UI. But [Gcode Simulator](https://play.google.com/store/apps/details?id=de.dietzm.gcodesimulator&hl=ru) and Pronterface shows it would be filled.
Another problem with the model and maybe the root problem is The Cura shows filled slice in couple of levels.

daid, 2016-02-29:
Look in x-ray view, it will show that your model has a lot of holes. Cura will do it's best in this case, but can get confused.

closed 2016-02-29 by daid

iHateInventNames, 2016-03-01:
Slic3r makes one hole without support. Cura fills the hole inside. Try 100%
Slic3r losts one gear though

iHateInventNames, 2016-03-01:
Tried on latest stable version for windows

daid, 2016-03-01:
Like I said, the model is broken, results will be unpredictable in any software.

iHateInventNames, 2016-03-01:
What do you mean by broken? How you see it? OpenCascad shows good model.
Cura shows good model. But gcode it generates has problems.

BagelOrb, 2016-03-01:
Look at the x-ray view. If there are red parts your model is missing some
faces, or it has too many.

Try playing with the fix horrible settings in advanced. Maybe some
combination can fix your problem. Otherwise you have to fix the model
itself.

Openscad doesn't handle it very well when it performs a difference or a
union of two object which have a face in the same plane...

iHateInventNames, 2016-03-01:
Maybe I missing something, but any problems of STL should be seen in Cura, but:
What I see in Cura is:
[image]
and what I see after printing is:
[image]
again, Cura:
[image]
[image]
[image]
[image]
[image]
looks like it creates some solid part, but it is not. well, it should not:
[image]
[image]
stlview.com shows good model too:
[image]
What is hex ray view? Of Cura. Everything well there too.
I would be gratefully thankful for any proposes.

daid, 2016-03-01:
[image]

nallath, 2016-03-01:
If there are red areas, your model is wrong, regardless of if your model is displayed correctly. There are red areas, so your model is non-manifold (http://blender.stackexchange.com/questions/7910/what-is-non-manifold-geometry)

iHateInventNames, 2016-03-01:
You mean these intersection areas? Its not a problem to Cura. It was just bit esier to write drawing script. What problem is Where Cura those faces took from? There is no in STL. There should be a hole like Slic3r makes.

nallath, 2016-03-02:
Cura took those faces from the model file. Those faces are there in the STL. There are a number of ways a bit of software can handle this, but not all of them work in every situation. Slic3r probably has different defaults from cura. Try messing with the fix horrible options in expert mode (or even better, fix the model)

Ghostkeeper, 2016-03-02:
The problem essentially comes from that it can't determine what the inside of the model is, so it doesn't know what to fill. Like if you have a house without a roof on top, everything inside the house becomes "outside".

In iHateInventNames' case, it looks like there are extra surfaces inside the model. The program then doesn't know what faces to use. If it chooses the extra surfaces your base becomes round and the teeth would be assumed to be non-watertight or "outside". If it chooses the teeth then only the teeth would appear and the base would become non-watertight or "outside". So what happens varies from program to program. In Cura's case it was decided that there would be no programming to account for that case at all. Therefore the result is unpredictable.

The extra surfaces are, of course, only visible in Normal view if you move the camera inside your model. Since that is a hassle, there is also the X-Ray view where you can more easily detect such cases: It colours them red.

iHateInventNames, 2016-03-02:
If I'd eliminate these issues would someone look onto the problem? If yes, I'll send updated model tomorrow.

iHateInventNames, 2016-03-07:
Thank you, that helped. Sory for bothering you guys. I fixed model with 3d builder automaticaly. Can we hope Cura could do that in feature?

BagelOrb, 2016-03-08:
Let's hope so.