daid/LegacyCura #1424: Problem open setting window at the first time using Cura for MAC OS El Capitano

mhnguyen233, 2016-03-02:
Hello everyone,

When installing Cura dev version , i have a problem at the first time using Cura . I cannot open Expert Setting window or Preference windows after notice "Welcome to the new version", it blocks there. However , when i click on Load File , that windows can appear. After, when i quit program and reopen , every windows open normally.
This problem appears only when the first time ( when we don't have any profile and preference on machine). 
I don't know if you guys have a similar problem and someone may help me resolve it ?

Thank you in advance and sorry for my bad englih

daid, 2016-03-02:
This is a known bug in OSX. We do not have any workaround or fix that we could find, and is related to the wxWidgets toolkit that we are using.

mhnguyen233, 2016-03-03:
Hello @daid , 

Thank you for your reply , yesterday i have tried some method to fix that bug and finally i have one to resolve it , using function "wx.CallAfter(self.StupidMacOSWorkaround)" to appear the panel and function"ToggleWindowStyle(wx.STAY_ON_TOP)" to brint it in front of windows. 
My version mac works without problem now !!