daid/LegacyCura #1425: lost settings

meme9999, 2016-03-03:
Every time I update to the latest version of Cura all my previous settings are lost.

NachoValdesCarriles, 2016-04-07:
I got the same problem updating cura on Mac OS X 10. I lost everything, my 3 printer profiles, and now I have to configure everything again. And I don't remember the value that I configured for some speed settings, so I have to try differents values again. Shit

BagelOrb, 2016-04-07:
One thing you could do is load back your settings from a gcode..

daid, 2016-04-08:
Note that old Cura (15.04.x) stores it's settings per version in ~/Library/Application Support/Cura/[version]
So you can copy older versions into newer version if you want (and when the automatic copy fails)