daid/LegacyCura #1426: Excess slots in print

glen-cafferty, 2016-03-07:
I have a part that I made by merging several other parts in 123D Design. It looks OK in 123D and Meshmixer and on the print platform view in Cura. When  printing (and in the layer view) there are slots not shown on the other views of the model in the first and last few layers. These appear to be where parts were joined together in the design. Tried to attach the STL file but your site doesn't recognize that format. attached are screen shots, the problem is in the lower right of the image. Solutions please! 
[image]
[image]

daid, 2016-03-07:
How does the model look in X-Ray view? And red parts?

glen-cafferty, 2016-03-09:
daid, 

None. It also looks good in 123D Design and Meshmixer. I'm redesigning the part, but would still like to know what the problem is. 

Glen 

----- Original Message -----

How does the model look in X-Ray view? Any red parts?

BagelOrb, 2016-03-30:
Try fiddling around with the fir horrible settings in expert settings...

glen-cafferty, 2016-03-31:
Tim, 

Thanks, that did it. 

Glen

closed 2016-03-31 by nallath