daid/LegacyCura #1427: Y and Z scale factors swapped when scaling non-uniformally

mastory, 2016-03-07:
A scale factor entered into the Y factor changes the Z scale.  The converse is also true.

Scaling related question; What is the need or purposed for 'Snap Scaling'?  For me it has only caused problems when I forgot to uncheck the box.

daid, 2016-03-07:
I think this issue belongs at https://github.com/Ultimaker/Cura/issues
(I know, confusing. Sorry)

Ghostkeeper, 2016-03-07:
Tested both on Cura 2.0.99 and on 15.04 and couldn't reproduce this. 1.99.0 though is a fairly old prototype of 2.0.99. Try the newer one at https://software.ultimaker.com/Cura_closed_beta/2.1/

mastory, 2016-03-07:
I havnt had access to the latest beta until now.  I'll try it in that ver, and check back.

Apparently I've posted in the wrong issues location per Daid.  If I see the same thing in the newer beta, I move the issue.  Otherwise I'll clear it.

Thanks

<div>-------- Original message --------</div><div>From: Ghostkeeper <notifications@github.com> </div><div>Date:03/07/2016  3:50 AM  (GMT-05:00) </div><div>To: daid/Cura <Cura@noreply.github.com> </div><div>Cc: mastory <matthew_a_story@yahoo.com> </div><div>Subject: Re: [Cura] Y and Z scale factors swapped when scaling non-uniformally  (#1427) </div><div>
</div>Tested both on Cura 2.0.99 and on 15.04 and couldn't reproduce this. 1.99.0 though is a fairly old prototype of 2.0.99. Try the newer one at https://software.ultimaker.com/Cura_closed_beta/2.1/

mastory, 2016-03-10:
I installed the newest 2.0.99, and there are similar problems, not the one I described above.  I will report those in the correct place, and close this issue.  Sorry for the confusion.

closed 2016-03-10 by mastory