daid/LegacyCura #1429: Mode position in gcode differs from shown

iHateInventNames, 2016-03-07:


BagelOrb, 2016-03-07:
Can you elaborate?

iHateInventNames, 2016-03-08:
filled its description

BagelOrb, 2016-03-08:
If I slice this in Cura 15.04.4 and switch between normal and layer view I see the gcode at the same position as the model.

Please fill in a complete bug description. You supply far too less information.

iHateInventNames, 2016-03-08:
Normal and Layer view in Cura shows everything well. But once I load the gcode into GCode simulator or Pronterface, I see it shifted and upper layers became out of borders.

BagelOrb, 2016-03-09:
Sounds like a problem of Pronterface.. Try reporting a bug with them..

iHateInventNames, 2016-03-09:
oh well, the problem in 3d printer persists

BagelOrb, 2016-03-09:
What kind of printer do you have?

iHateInventNames, 2016-03-09:
reported bug to GCode simulator

closed 2016-03-09 by iHateInventNames

iHateInventNames, 2016-03-11:
I have smartcore 3d printer. Has no problem while printing Cura latest
official release made gcode with printrun. But the Cura itself sends gcode
shifted to 3d printer