daid/LegacyCura #1431: Cura will not build supports where the support is needed

Rdrost17, 2016-03-10:
I am trying to print a specific file that has an overhang on both sides about 0.125in from the base. One side will build supports the other will not. I have adjusted the support angles and everything but cannot make the software create supports. Any ideas?

BagelOrb, 2016-03-10:
This problem is addressed in the next big Cura release.

Rdrost17, 2016-03-10:
Do you know when this will be released?

BagelOrb, 2016-03-11:
Could be a couple of weeks, could be a couple of months...

Ghostkeeper, 2016-03-11:
Closed beta is going on right now. Public beta will be soon. Release is currently set for the end of the month.

fiberbug, 2016-05-03:
I am having similar problems trying to print a spool and asking for supports? If I just print one side of the spool, the supports work good but when I try to print the entire spool and ask for supports, there are none?

BagelOrb, 2016-05-03:
Try out the new beta release! The support system has had a total overhaul.
Now it will place support in many places it wouldn't before.

fiberbug, 2016-05-03:
Where do I go to get the New beta release?

fiberbug, 2016-05-03:
> Tim Kuipers <mailto:notifications@github.com>
> Tuesday, May 03, 2016 3:42 PM

Hi Tim,

Thanks for the response. Could you tell me how or where I should go to 
try the new beta release?

nallath, 2016-05-04:
http://software.ultimaker.com/Cura_open_beta/

Ghostkeeper, 2016-05-04:
You can get it from [the Ultimaker website](https://ultimaker.com/en/products/cura-software) with the smaller text below the main download button ("Download Cura 2.1.0 (beta)"). Or just download it from our [open beta page](http://software.ultimaker.com/Cura_open_beta).