daid/LegacyCura #1432: How to install on OpenSuse

alexpacini, 2016-03-14:
Hi,
I am trying to install the software on OpenSuse Leap42.1.
Unfortunately, the official package is only for Debian based systems and the unofficial OpenSuse build (OBS) has issues with the wxWidget version and it is not updated to the last version.
Is it possible to get instructions on how to build the package on OpenSuse?
Why the official version is distributed only as .deb?

Thanks,
Alex

XiaoShiZi, 2016-03-14:
http://software.opensuse.org/package/Cura

Checked that site?
Greetz, Xiao Shi Zi

alexpacini, 2016-03-14:
Yes, but the latest package available there (15.04) does not work on Leap 42.1

daid, 2016-03-14:
> Why the official version is distributed only as .deb?

Because that's all we care about. We do not have the man power to support all Linux distros. So we have a debian package, and sources.

closed 2016-03-14 by daid

alexpacini, 2016-03-21:
Why don't ufficially support a Docker image?

daid, 2016-03-22:
effort.