daid/LegacyCura #1434: Failed to save

alxbirdsall, 2016-03-15:
So after a couple weeks of using my Ultimaker 2+ and Cura. I have encountered my first problem. Suddenly every 3d model I try and save to SD in cura comes up with the message 'failed to save'. No explanation as to why. So now I am very stuck, couldn't find any answers online. I have tried uninstalling and reinstalling cura, and tried a variety of different SD cards.

daid, 2016-03-15:
Does coping a file to the SD card work?
Which OS are you using?

Ghostkeeper, 2016-03-16:
Also always make sure that the SD cards are not on "locked" :wink:

mihaidimitras, 2016-05-22:
I have the same problem I checked the sd card is on unlock the same result 'failed to save'????

Ghostkeeper, 2016-05-23:
Just to check, same question for you: Is it possible to copy files to the SD card via the normal file explorer?

mihaidimitras, 2016-05-23:
No

Ghostkeeper, 2016-05-23:
Then the problem lies not with Cura, but with the SD card or your computer. Perhaps the computer doesn't support the file format on the SD card? You might need to format it.

closed 2016-05-23 by daid

mihaidimitras, 2016-05-23:
I think it could be related to admin permission.  Anyway thanks...