daid/LegacyCura #1435: Rotation of Part does not upgrade Slicing Cura 15.4.4 Windows/Mac

mruepp, 2016-03-15:
With Cura 15.4.4 when placing an Object and rotate it, the slicing stays with the original import orientation. Cura starts to print the Object without rotation.

daid, 2016-03-15:
Does work when I try it. Does this happen with a specific 3D model? or printer type?

mruepp, 2016-03-15:
No, every Model i import on Mac and Windows. I tried with Host Mode (directconnect via USB) and by exporting GCode. I import the Model, then rotate it to fit on the Printbed and when I print it prints in the initial import orientation. Printer is a Velleman K8200 with and without K8204 Z-Axis Upgrade with and without direct vs SD Card Communication (K8201).

daid, 2016-03-15:
Ok. Where did you download this version from? Do you see a new "progress bar" below the save button when you rotate the model?

mruepp, 2016-03-15:
This is the download link https://ultimaker.com/en/products/cura-software 
Also I recognize a short progressbar in the 3D windows upper left corner after rotating.

daid, 2016-03-17:
Odd, could you get a log from the OSX version. See https://github.com/daid/Cura/wiki/MacOS---Getting-a-proper-log

mruepp, 2016-03-17:
Hi, here is the log. However I did a rotating/slicing save Gcode and reimported it on Repetier and on Cura (on the Mac Version) and it now seems to be ok.

I will check on the Windows Version if I have time to (am on holidays for the next two weeks)

sunrise:MacOS michael$ ./Cura 
load preferences from /Users/michael/Library/Application Support/Cura/15.04.4/preferences.ini
2016-03-17 14:45:25.310 Cura[3129:157622] GLUT Warning: glutInit being called a second time.
Listening for engine communications on 49674
Traceback (most recent call last):
  File "Cura/util/sliceEngine.pyo", line 169, in submitInfoOnline
  File "urllib2.pyo", line 126, in urlopen
  File "urllib2.pyo", line 400, in open
  File "urllib2.pyo", line 418, in _open
  File "urllib2.pyo", line 378, in _call_chain
  File "urllib2.pyo", line 1215, in https_open
  File "urllib2.pyo", line 1177, in do_open
URLError: <urlopen error [Errno 61] Connection refused>
Closing down
Shader was not properly released!
Shader was not properly released!
Shader was not properly released!

closed 2018-12-14 by mruepp