daid/LegacyCura #1436: If you cancel print (via USB) fans, bed and extruder still powered

tiwanacote, 2016-03-17:
I tried to use:

In printWindow.py I tried this but didn´t work. How could I send Gcode to power off devices when you cancel a print?

def OnCancel(self, e):
       self._printerConnection.sendCommand("M107")
       self._printerConnection.sendCommand("M190 S0")	
       self._printerConnection.sendCommand("M109 S0")
       self._printerConnection.cancelPrint()

Thankyou!

tiwanacote, 2016-03-18:
Ok...   this is not an elegant way but works fine

in printWindow.py:

	def OnCancel(self, e):	
		self._printerConnection.cancelPrint()
		self._printerConnection.openActiveConnection()
		self._printerConnection.sendCommand('M107')
		self._printerConnection.sendCommand('M190 S0')
		self._printerConnection.sendCommand('M109 S0')
		self._printerConnection.closeActiveConnection()

Ghostkeeper, 2016-03-21:
What printer was this on?

tiwanacote, 2016-03-22:
Sorry, good question...

I found it in Repraps setting machine (I left my printer powered all weekend long). May be end.gcode must be sent if "cancel print" button is pressed.

Thankyou

Ghostkeeper, 2016-03-23:
Yeah, that might be necessary. Currently it sends (in printWindow.py:620):

    M104 S0
    M140 S0

This turns off the heating of the nozzle and the heated bed, if I'm not mistaken. Calling the ending g-code at cancel would be a good solution, I think. This should account also for printers that like to decide for themselves when to turn the fans (or the heaters) off after a print.