daid/LegacyCura #1438: Can not run Cura through ssh with X forwarding or with X2Go

alexpacini, 2016-03-21:
I installed cura on a mint 17.3 xfce machine. 
While it works as expected if logging in regularly into that machine, I can not have it working using ssh with X forwarding or using X2Go.
Cura is the only non working application.
To be precise, the 3d view is only part missing. The outer interface is present, but unable to import any object.

daid, 2016-03-22:
No clue. Cura is just using openGL to render that view. Does glxgears work trough your remote X?

alexpacini, 2016-03-22:
glxgears works as expected.

The output launching from the cli:
$ cura
load preferences from /home/alex/.cura/15.04.4/preferences.ini
Listening for engine communications on 49674

If I click on the 3D plane (though not visible) i get:

Traceback (most recent call last):
  File "/usr/share/cura/Cura/gui/util/openglGui.py", line 187, in
_OnGuiMouseDown
    self.OnMouseDown(e)
  File "/usr/share/cura/Cura/gui/sceneView.py", line 835, in OnMouseDown
    self._mouseClick3DPos = self._mouse3Dpos
AttributeError: 'SceneView' object has no attribute '_mouse3Dpos'
Traceback (most recent call last):
  File "/usr/share/cura/Cura/gui/util/openglGui.py", line 187, in
_OnGuiMouseDown
    self.OnMouseDown(e)
  File "/usr/share/cura/Cura/gui/sceneView.py", line 835, in OnMouseDown
    self._mouseClick3DPos = self._mouse3Dpos
AttributeError: 'SceneView' object has no attribute '_mouse3Dpos'
Traceback (most recent call last):
  File "/usr/share/cura/Cura/gui/util/openglGui.py", line 187, in
_OnGuiMouseDown
    self.OnMouseDown(e)
  File "/usr/share/cura/Cura/gui/sceneView.py", line 835, in OnMouseDown
    self._mouseClick3DPos = self._mouse3Dpos
AttributeError: 'SceneView' object has no attribute '_mouse3Dpos'
Closing down
Segmentation fault

---
Alex Pacini, MS
email: alexpacini90@gmail.com
blog: https://pacinispace.blogspot.com
ORCID: orcid.org/0000-0002-0316-4989
skype: alex_pacini

---

alexpacini, 2016-03-22:
Just to add an info: it is a fresh system. An updated Mint 17.3 Xfce with only cura installed.
Using the same system inside a VirtualBox in the remote machine everything works (tried with X2Go).

daid, 2016-03-22:
Then I have no clue.

The backtrace you see is as expected, as the _mouse3Dpos normally gets set by the 3D rendering code (which uses some tricks of reading back drawing buffers to define where the mouse is located in 3D space)

alexpacini, 2016-03-23:
The beta version (Qt based) works as expected. Unfortunately the UM2+ is not listed there (in the last version: 15.06.3) and it is sometimes a bit buggy and prone to crash.

daid, 2016-03-23:
Send me an email, I'll enlist you in the latest Cura beta. The 15.06 is quite old and buggy compared to that.
d.braam@ultimaker.com