daid/LegacyCura #1439: Cura 15.04.4 Does not 'Save Model' on Ubuntu 15.10

WVUTampaAlum, 2016-03-23:
I have tried multiple times but I am unable to save any model file from cura to internal file system or sd card.

cura 15.04.4
ubuntu 15.10

Ghostkeeper, 2016-03-24:
Are you getting any errors?

WVUTampaAlum, 2016-03-24:
No errors and no message at all like when I save gcode or profile.  Is there a log location I can look for more details?

Ghostkeeper, 2016-03-24:
Depends on your platform:
* On Windows, the log is by default written to `C:/Users/<user>/AppData/.cura` somewhere.
* On Linux, you can get logs by running Cura from a command line: python cura.py (You may have to browse the command line to the source directory.)
* On OSX, read instructions here: https://github.com/daid/Cura/wiki/MacOS---Getting-a-proper-log

WVUTampaAlum, 2016-03-24:
That helped and I figured out my issue.  I don't know if this is a bug or as designed.

The error was.  Error: Unknown model extension:

I was only putting a file name in the save dialog without extension.  When I saw the error I tried putting either .amf and .stl and it saved for both.  

I would say either automatically add one of the file types to the file when saving or at a minimum display a message on the screen that file extension is required.

Thanks!

daid, 2016-03-24:
Ah, yes, this is a linux specific problem. Both windows and OSX automatically add the extension in the save dialog.