daid/LegacyCura #1440: Prusa i3 drags Y while printing

patrickrottman, 2016-03-24:
Its like during prints my Y coordinants gets pulled while its printing. I have to use a different program now because nothing I print is working. It used to work wonderfully until I updated to the latest version

BagelOrb, 2016-03-24:
Thats weird I cannot imagine the cause is in Cura. Did you find that other
slicers didn't have that problem? It sounds like a hardware problem to me..

patrickrottman, 2016-03-24:
When I started using "slic3r" everything started working again. If results were inconsistent like if when using Cura every 5th one got messed up I would think it was hardware, but every single export from Cura now does this, and every one from Slic3r works first try. 

I'm wondering if it could be my machine profile? None of the normal settings could cause this but maybe the flavor I'm using? I have found a solution by just not using Cura now but it is by far the best interface for G-codifying.

daid, 2016-03-24:
Could be your firmware settings. Cura does no effort in trying to limit Z speeds, as it's our opinion that that's the firmware task. Slic3r however, can limit the Z speed.

patrickrottman, 2016-03-24:
Got it! Not Cura, it was me:(
Cura is still amazing, my Y axis belt had stretched causing it to jump occasionally. The settings on slic3r were slower than I had them set in Cura meaning it didn't happen for that one. 

Sorry for wasting your time!

closed 2023-05-13 by patrickrottman