daid/LegacyCura #1442: gcode convertion failure

klooppjk, 2016-03-25:
I load an .obj file
preview is correct into cura
if i select "slice view" 
the hole in the midle disapears the problem exists when I print the object !
[bouchon_bas.zip](https://github.com/daid/Cura/files/189329/bouchon_bas.zip)

klooppjk, 2016-03-25:
i have already  tried to uncheck the "fix horrible" it does not solve the issue

daid, 2016-03-25:
Did you check the X-Ray view? If it shows any red parts, then there are errors in the model.

klooppjk, 2016-03-25:
there are no "red parts", you can open the zip file I sent if you want.

BagelOrb, 2016-03-26:
Try fiddling with the fix horrible settings under export settings.

klooppjk, 2016-03-29:
Thanks for your help.
I have tried to check and uncheck "fix horrible" setting options without success. (i have tried ALL possibilities !)

daid, 2016-03-29:
[image]
There are a shitloads of holes in that model. Cura cannot make up information that is not there.

closed 2016-03-29 by daid