daid/LegacyCura #1443: Anfahrweg

Timechild2013, 2016-03-30:
Hello, my englisch is not very well, so I write in German.
Hallo, ich vermisse die Einstellung mit dem Anfahrweg bzw Abfahrweg. um die Wände frei von Punkten, wenn der Druckkopf auf Anfangsdruckpunkt aufsetzt, zu bekommen. Nach dem Drucken bekomme ich lauter häßchliche Püncktchen auf der Fläche zu sehen. Um das verständlicher zu erklären ist das vergleichbar mit Fräsen. Bevor der Fräser in das Material eintaucht, macht er einen Halbkreis, Viertelkreis oder eine Schrägeanfahrt, um einen sauberen Übergang zu ermöglichen.

BagelOrb, 2016-03-30:
I don't understand why you would make a small move before starting an
extrusion move. I don't see how that would reduce the blob at the beginning
of a printed line which results from building up the Bowden tube pressure.

One approach would be to start moving slowly and speed up until the Bowden
pressure is the required amount.

Some people like to use coasting to reduce the 'delay' effects of the
Bowden tube, but it isn't totally adequate for it as it is currently
implemented in Cura 2.

I hope I understand you correctly...

Timechild2013, 2016-03-30:
I mean this one.
[image]
[image]

Timechild2013, 2016-05-04:
Hello, thanks for your help with Cura2.
I can fix my problem, but I cant not fix this 2nd problem.
In Cura2 I dont have choice to tell it print outer or inside line first, only infill or wall first.
My problem is, if the area or cirle have a big angle, so the outer wall/line fall before inside line printed if have select 2 line as wall. I can use support line, but this option is not perfect.

[image]

closed 2016-05-05 by Timechild2013